r2axz/bluepill-serial-monster #10: .inf file

TadasLes, 2020-11-20:
Tested on W7 and W8. It seems, driver .inf file is necessary to extend usable OS range.
Thank you
Tadas LY1CE

r2axz, 2020-11-20:
Thank you for your report! I haven't tested the firmware on those. I should have, though. You right, an inf file is needed to make it work. I will address this issue shortly.

r2axz, 2020-11-20:
@TadasLes Would you mind testing the inf file when it is ready. I have a Windows 7 VM and I'm going use it for testing, but it would be nice if you can confirm the inf file works on Windows 8 as well.

TadasLes, 2020-11-20:
> @TadasLes Would you mind testing the inf file when it is ready. I have a Windows 7 VM and I'm going use it for testing, but it would be nice if you can confirm the inf file works on Windows 8 as well.

Of course I will test it on W7, W8 and W XP

r2axz, 2020-11-20:
@TadasLes Thank you! I really appreciate that. I think the inf file will be available either later today, or during the weekend.

r2axz, 2020-11-20:
@TadasLes Looks like I managed to create an INF file that works just fine on Windows 7 x64. Could you please test it on your set of operating systems?

Here is how I do it on Windows 7 x64:

Method 1

1. Copy both inf and cat files to your machine;
2. Open the Command Prompt with administrative privileges;
3. Run the following command:

```cmd
pnputil -i -a stm32-blue-pill-serial-monster.inf
```

It should warn you that you are installing a driver from an unidentified developer. That's OK, the driver has been signed with a self-signed cert.

4. Plug in the device. Windows will first try to find the driver in Windows Update, and after that will install the local driver you just installed.

Method 2

Go to the Device Manager, right click on one of the STM32 Blue Pill Serial Monster devices, and select Update Driver Software. Then choose browse my computer for driver software and point it to a directory with the inf file. Windows will install the driver for one of the CDC devices. You don't need to repeat the procedure for the others. Just disconnect and reconnect the device and Windows will pick up the rest of the ports.

See below for the inf file archive.

[inf.zip](https://github.com/r2axz/bluepill-serial-monster/files/5576273/inf.zip)

r2axz, 2020-11-21:
@TadasLes I tested on Windows XP 32 bit myself. Everything's OK.

r2axz, 2020-11-21:
I bought a code-signing certificate from Comodo. As soon as they complete their validation stuff, I will sign the inf.

TadasLes, 2020-11-21:
XP x32 and W7 x32 - OK
W8 x64 - the system interrupts installation without valid driver certificate. There isn't branch for installing non certified driver.

TadasLes, 2020-11-21:
Hope, with certified driver things will be OK.

r2axz, 2020-11-21:
@TadasLes Thank you very much for testing. I will get back to you when my certificate is ready.

If you don't mind, what's your use case for the firmware?

r2axz, 2020-11-23:
@TadasLes I added signed cat file. Now it should work on Win 8 as well. Could you please re-test?
[inf.zip](https://github.com/r2axz/bluepill-serial-monster/files/5582573/inf.zip)

r2axz, 2020-11-23:
I also testing it on Win 8 x64 myself. Now it works OK.

TadasLes, 2020-11-23:
Hi, tested signed driver on W8 x64. Install successful.
I'm planning add to BluePill level shifters on lines that are not 5V tolerant for safety.
It will be fine tool for debugging systems with UART. No more single USB <-> UART converters.
:1st_place_medal:

r2axz, 2020-11-23:
@TadasLes Thank you very much for testing. And thank you for describing the use-case, that's interesting. Looks like we're all set, I am closing the issue and merging the pull-request.

closed 2020-11-23 by r2axz