r2axz/bluepill-serial-monster #24: [FR] Invert RX TX

zbyrek, 2021-02-06:
Hi,
is it possible to add to configuration option to invert RX and TX or there is hardware limitation?

r2axz, 2021-02-07:
Hi @zbyrek,

unfortunately, it is impossible to invert RX, TX, or CTS due to the MCU hardware limitations.

What it is that you are trying to achieve? I am going to make a wild guess, but if you are connecting bluebill-serial-monster to RS-232, inverting signals is not the way to go. You need to use some RS-232 level converter circuitry.

Please keep in mind that it is still possible to change TX output type (push-pull vs open drain), and RX/CTS pull type (up, down, floating). This can be useful when designing a level conversion circuit.

zbyrek, 2021-02-07:
I know that rs232 is another voltage levels, just simplest way to change level is using single transistor which inverts signal, also i thought about ws2812 over uart.

r2axz, 2021-02-08:
@zbyrek, thanks for the explanation. Since it is impossible to change signal polarity of RX/TX due to hardware limitations and the fact that this is documented in README.md, I am going to close this issue. Please use hardware solutions. As I said, keep in mind that you can configure TX as an open-drain output. If you do that and use a pull-up resistor, you will effectively invert TX. Thanks.

closed 2021-02-08 by r2axz