r2axz/bluepill-serial-monster #26: Non-free library dependency

zoobab, 2021-03-01:
Hi,

This GPLed project depends on a non-free library (SMT32CubeF1), where the USB stack is licensed under the ST SLA0044:

https://github.com/STMicroelectronics/STM32CubeF1/blob/master/License.md

Like we did for the DirtyJTAG project, you could use free libraries like the unicore-mx or libopencm3.

I could try to change the underlying library.

r2axz, 2021-03-01:
I don't use the USB stack from SMT32CubeF1, this project implements the entire USB stack itself.

r2axz, 2021-03-01:
@zoobab This project uses CMSIS only, which I believe is open. CMSIS uses Apache license so there is no any problems.

zoobab, 2021-03-01:
OK, if you only use the CMSIS library from ARM, is it possible to modify the build instructions to only use CMSIS? 

https://github.com/ARM-software/CMSIS_5

r2axz, 2021-03-01:
Sounds reasonable, thank you. I will implement this at some point.

zoobab, 2021-03-01:
I removed everything except "STM32CubeF1/Drivers/CMSIS", and it still builds. Will try with CMSIS_5 now...

r2axz, 2021-03-01:
Thank you for this.

zoobab, 2021-03-02:
Ideally, via a "git submodule", although after 10 years this feature of git still sucks.

r2axz, 2021-03-02:
You probably right. I will do this. Most likely during the weekend. Pretty busy with my work-work at the moment. Thanks!

r2axz, 2021-04-30:
Turns out https://github.com/ARM-software/CMSIS_5 is not enough. bluebill-serial-monster uses vendor-specific CMSIS definitions from STM. However, those definitions are distributed under BSD 3-Clause license so there is no non-free library dependencies in the project. Again, I don't use the STM USB stack. Closing the issue.

closed 2021-04-30 by r2axz