r2axz/bluepill-serial-monster #33: Recommended version of gcc-arm-none-eabi for building FW

iDoka, 2021-07-13:
What is your experience to building project with various versions of gcc-arm-none-eabi?

I tried `gcc-arm-none-eabi-10-2020-q4-major` and have some warnings:
```
usb_cdc.c: In function 'usb_cdc_port_start_tx':
usb_cdc.c:447:41: warning: array subscript '<unknown>' is outside the bounds of an interior zero-length array 'uint8_t[0]' {aka 'unsigned char[]'} [-Wzero-length-bounds]
  447 |             dma_tx_ch->CMAR = (uint32_t)&tx_buf->data[tx_buf->tail];
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from usb_cdc.c:10:
circ_buf.h:15:13: note: while referencing 'data'
   15 |     uint8_t data[0];
      |             ^~~~
usb_cdc.c: In function 'DMA1_Channel4_IRQHandler':
usb_cdc.c:447:41: warning: array subscript '<unknown>' is outside the bounds of an interior zero-length array 'uint8_t[0]' {aka 'unsigned char[]'} [-Wzero-length-bounds]
  447 |             dma_tx_ch->CMAR = (uint32_t)&tx_buf->data[tx_buf->tail];
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from usb_cdc.c:10:
circ_buf.h:15:13: note: while referencing 'data'
   15 |     uint8_t data[0];
      |             ^~~~
usb_cdc.c: In function 'DMA1_Channel7_IRQHandler':
usb_cdc.c:447:41: warning: array subscript '<unknown>' is outside the bounds of an interior zero-length array 'uint8_t[0]' {aka 'unsigned char[]'} [-Wzero-length-bounds]
  447 |             dma_tx_ch->CMAR = (uint32_t)&tx_buf->data[tx_buf->tail];
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from usb_cdc.c:10:
circ_buf.h:15:13: note: while referencing 'data'
   15 |     uint8_t data[0];
      |             ^~~~
usb_cdc.c: In function 'DMA1_Channel2_IRQHandler':
usb_cdc.c:447:41: warning: array subscript '<unknown>' is outside the bounds of an interior zero-length array 'uint8_t[0]' {aka 'unsigned char[]'} [-Wzero-length-bounds]
  447 |             dma_tx_ch->CMAR = (uint32_t)&tx_buf->data[tx_buf->tail];
      |                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from usb_cdc.c:10:
circ_buf.h:15:13: note: while referencing 'data'
   15 |     uint8_t data[0];
      |             ^~~~
usb_cdc.c: In function 'usb_cdc_poll':
usb_cdc.c:331:36: warning: array subscript '<unknown>' is outside the bounds of an interior zero-length array 'uint8_t[0]' {aka 'unsigned char[]'} [-Wzero-length-bounds]
  331 |                 uint8_t *buf_ptr = &rx_buf->data[rx_buf->tail];
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from usb_cdc.c:10:
circ_buf.h:15:13: note: while referencing 'data'
   15 |     uint8_t data[0];
      |             ^~~~

```

[image]

Maybe exists version of gcc-arm-none-eabi which works fine w/o these warnings?

Nable80, 2021-07-13:
Don't blame your compiler for these warnings, please.
It does the right thing™ and it's better to fix the code instead (``data[]`` instead of ``data[0]``).

r2axz, 2021-07-14:
Fair point, ```data[]``` is by any means the cleanest way to define flexible size arrays. Please test the linked PR.

As for the compiler version, I use arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release) on my Mac which I use for development, and gcc-arm-none-eabi_15%3a9-2019-q4-0ubuntu1_amd64, which is used  in GitHub actions. None of them shows this warning.

iDoka, 2021-07-14:
I checked your PR.
Now `gcc-arm-none-eabi-10-2020-q4-major` works fine without any warnings.

Thanks for your quick fix!

closed 2021-07-14 by r2axz