r2axz/bluepill-serial-monster #34: LED indication enchancement

iDoka, 2021-07-13:
My 50 cents for improve User eXperience with Serial Monster:
As I can see in current moment Serial Monster used single LED to tell us any activity on UART ports (TX&RX).

I find very useful to use the second LED to help us to separating activity indication, e.g.:

* LED1 - any TX activity on UART ports
* LED2 - any RX activity on UART ports

To make accessability of second LED (by default use as POWER LED) we should move corresponding resistor.

So, LED1 already available on PC13 pin, and LED2 will be available on PC14 pin (in proposed setup).

Here is my example to moving resistor: 

[image]

[image]

Hope this HW mod will be easy and useful for most Serial Monster's users.

r2axz, 2021-07-14:
In fact, currently, the LED blinks on any USB activity, not only RX/TX events. This includes control and interrupt endpoints events as well. For instance, when some control line changes its state (say RI) and this state is reported to the host, the LED blinks. Also, the LED blinks during the device configuration phase, serial port configuration changes, etc.

Your proposal seems to be focused only on TX/RX events, dropping everything else.

Also, can you provide an example use case where separate TX/RX indication improves the user experience?

marinjurjevic, 2021-07-14:
Hi guys,

I would not opt-in reusing PWR LED because it would introduce unnecessary change. PWR led is like a hardware sanity check, if it's not lit, something is wrong with the way board is being used. Same thing for PC13, I like information that there is activity on the USB lines.

The described problem should be left for end user to solve by adding additional hardware which can do the complete job -- tracking activity on any line he wishes.

I discovered this project few days ago and I absolutely love the job being done here, so that's my two cents this change.

iDoka, 2021-07-14:
Guys, thanks for your comments!

I borrow this idea from another project: https://github.com/candle-usb/candleLight_fw
It's FW for USB-to-CAN dongle based on STM32.
This dongle have only two LEDs with following control logic:
* When connected to USB the LEDs blinking once, and then off
* When interface (e.g. can0) bringed up the all LEDs is on
* When CAN TX event occurs the LED1 blow out for a short time (like "blinking")
* When CAN RX event occurs the LED2 blow out for a short time (like "blinking")

> PWR led is like a hardware sanity check, if it's not lit, something is wrong with the way board is being used. 
This is a high price for the single function (only HW sanity check), It might be implemented in conjunction with enchanced LED functionality. 

> Your proposal seems to be focused only on TX/RX events, dropping everything else.
Not exactly, it was my initial mistake about LED control logic in Serial Monster.

> can you provide an example use case where separate TX/RX indication improves the user experience?
I think there are two basic scenarios: Interactive and Non-ineractive work.
When we work in Non-ineractive mode separating TX/RX blinking events might be useful to diagnostic SW/HW issues.
It help us to understanding where is fault node in our chain: DUT - Bluepill - Host

r2axz, 2021-09-28:
Unfortunately, I don't think I have time to implement this. Additionally, I am still not sure if this functionality is really needed and whether most users are willing to modify their boards.

Nevertheless, if you believe this functionality is needed, you can create a PR. Please make this conditional, whether through a build time flag or configuration setting. The PR should not affect the vanilla boards.

r2axz, 2022-02-08:
Closing for now, if somebody wants to add this functionality, please reopen.

closed 2022-02-08 by r2axz