r2axz/bluepill-serial-monster #36: Advanced option for Pull-Up resistor value changing

PufferFishStar, 2021-07-20:
Hi
Currently Pull-Up resistor is 1KΩ(?) and unchangeable, which I found is too small for receiving signals from some weak open-drain output.

In my case, I used the bluepill with another board with max 50mA output current. When the baud rate is 920k the bluepill can't receive message correctly. The current for switching from 0 to 1 is too big to finish ​in 1/920k second.

Finally I solved this issue by disabling the PU and put a external 10K PU.

thanks

r2axz, 2021-07-20:
Hi,

the PU/PD feature uses internal resistors of stm32f103c8t6. There is no way you can change their value programmatically. According to the datasheet, Rpu and Rpd are in the range of 30k to 50k, typically 40k, which is a weak PU/PD.

Unfortunately, this cannot be implemented.

PufferFishStar, 2021-07-21:
Got it, thanks!

closed 2021-07-21 by PufferFishStar