r2axz/bluepill-serial-monster #43: Проблема с работой с одним из устройств, может какие настройки?

steelpuxnastik, 2021-08-09:
Здравствуйте. Я вообще, на самом деле, не очень разбираюсь в теме работы с последовательным портом, определенно понимаю немного, потому может задаю тупой вопрос:
хочу использовать устройство для создания игрового устройства, которое использует 2 UART порта для связи с игрой Hatsune Miku Project Diva Arcade Future Tone и к Blue Pill подключается два устройсва, одно Arduino Micro, чтоб была возможность использования прерываний по последовательному порту, а второе плата на основе микроконтроллера Cypress cy8c4246azi-m445, с помощью которой работает сенсорная панель по последовательному порту (вот этот проект https://github.com/Project-Alpaca/LKP ) . Проблема в том, что по отдельности все проверенно работает, с STM32 Arduino работает без проблем, точно также как и плата сенсорной панели рабоатает без проблем с USB-TTL конвертером на основе CP2102, а вот при подключении к STM32 сенсорная панель определяется, но по сути не работает, ибо отзыв получается неверноятно медленно, раз в 5 секунд может быть. Я думал, что может быть проблема в UART, пробовал переключать между 1 и 3 портами (устройство работает от 5В, так что тут мне доступны только 1 и 3 порты, правильно?), также пробовал все предыдущие версии прошивок вплоть до 2.0, чтоб проверить, не временная ли это ошибка? Для подключения используется только RX/TX пины, ну и питание соответственно. Может доступные какие-то еще настройки или что-то в этом роде, потому что, как я уже писал, через CP2102 работает безукоризненно, потому не ожидал, что столкнусь с проблемами. На что обратить внимание? Заранее спасибо :)

r2axz, 2021-08-09:
Здравствуйте,

а какая скорость порта стоит на CP2102 и на serial monster? Точно все одинаково? Для подключения используется RX, TX, GND и VCC? Уровни сигналов точно правильные? Какой уровень сигнала на CP2102?

Далее, можно подключить один из свободных портов bluebill-serial-monster к RX/TX  между CP2102 и панелью и посмотреть как ходит трафик с помощью какого-нибудь эмулятора терминала (Tera Term, например), а затем переключиться на bluebill-serial-monster вместо CP2102, точно так же посмотреть трафик и посмотреть в чем разница. Только не забудьте RX соединять с TX, а TX c RX. То есть, крест-накрест.

Панель постоянно передает данные или пакетами?

steelpuxnastik, 2021-08-16:
Я поимел логический анализатор, мог бы скинуть результаты захвата. Единственное, что я из-за того, что сам не очень опытный пока не в состоянии понять, что там нужно. Самое интересное, что мне удалось определить, что каким-то образом по TX каналу в CP2102 почему-то ничего не передается, только в самом начале что-то чуть-чуть, хотя на самом деле должно передаваться постоянно что-то и очень много ибо с компьютера передается инфа на адресную светодиодную ленту как ей светится, ибо светится она постоянно и постоянно что-то меняется, но и на самом деле когда я подключаю светодиодную ленту то и все светится как положено и в функционале как бы не заложено определение подключена ли лента или нет - подключена по умолчанию всегда, RX канал при этом забит постоянно, как так, если я правильно понимаю, сенсорная панель постоянно опрашивается о своем состоянии, прикуснулись ли там к чему-то или нет. С bluebill-serial-monster такой проблемы вроде как не наблюдается, TX канал забит постоянно, также как и RX, но при этом все та же проблема, что все это работает очень медлено.
В каком формате лучше выгружать, если конечно стоит что-то выгружать?
На скриншотах может быть нифига не понятно, так как пробовал освоить все это, но в принципе наверное суть вышеописанного мною показывает. Если подскажете как более-менее правильно снять данные или на что обратить внимание, то естественно попробую это сделать :)
Ссылки на csv файлы экспорта, что попробовал сделать. 
https://drive.google.com/file/d/1KiO_nYTDymvE1g-qk1m7AUXnO4_D76ZE/view?usp=sharing
https://drive.google.com/file/d/1m1eqvPgRgF_nnDhcXicvEypnNOjEedCX/view?usp=sharing
[image]
[image]

r2axz, 2021-08-20:
А какая там все таки скорость обмена? Есть у меня идея почему такое может быть в теории. Но только если скорость обмена с панелью совсем небольшая.

r2axz, 2021-08-20:
Вернусь в понедельник, сделаю тестовую прошивку в отдельном брэнче. Сможете протестировать? Посмотрим, если поможет, то тогда буду думать что с этим делать глобально.

steelpuxnastik, 2021-08-20:
> А какая там все таки скорость обмена? Есть у меня идея почему такое может быть в теории. Но только если скорость обмена с панелью совсем небольшая.

А я так и не смог понять, на самом деле, какая там скорость, условно говоря он же пишет в каком промежутке по времени там сигналы шли, но не какая итоговая скорость обмена была. Все думаю, что сяду разобраться и все никак не получается снова замеры сделать. Но скорее всего завтра попробую еще.

> Вернусь в понедельник, сделаю тестовую прошивку в отдельном брэнче. Сможете протестировать? Посмотрим, если поможет, то тогда буду думать что с этим делать глобально.

Конечно смогу, спасибо большое :)

r2axz, 2021-08-23:
@steelpuxnastik пожалуйста, посмотрите решает ли проблему вот этот PR: https://github.com/r2axz/bluepill-serial-monster/pull/44
Собранная прошивка вот тут: https://github.com/r2axz/bluepill-serial-monster/suites/3571645963/artifacts/85796124

steelpuxnastik, 2021-08-25:
> @steelpuxnastik пожалуйста, посмотрите решает ли проблему вот этот PR: #44

Нет, к сожалению ничего не поменялось, все точно также :(

2.4.2 тоже попробовал - аналогично

r2axz, 2021-08-25:
Ничего страшного, PR все равно имеет смысл делать. А по поводу вашей проблемы я еще подумаю. Беру небольшой таймаут.

steelpuxnastik, 2021-09-01:
Таааак, наконец нашел в себе силы взяться включить анализатор, чтоб что-то посмотреть на счет него (меня просто немного голова напухала, когда я пытался все это собрать до того, чтоб запомнить, что я вообще куда тыкаю и что получаю в итоге, сейчас буквально пару пометок карандашом и многие моменты упростились для понимания :) Пометил, что **канал 0 - RX** TTL конвертера, **канал 1 - TX** TTL конвертера. На анализаторе выставил сканирование 8 MS/s, 
В целом, что заметил самое основное, что RX канал, вроде как на двух устройстах работает более-менее одинаково, то есть по времени появления сигналов и по времени их продолжительности вроде как более-менее сходно, намного большие вопросы вызывает TX канал, который вначале вроде как на bluebill-serial-monster кажется более быстрым, но в дальнейшем как-то все меняется наоборот, когда уже в период работы основного обмена на CP2102 всякие мелкие передачи 0xFF 125ns и именно в наносекундах, там быстро, а на bluebill-serial-monster я уже 0xFF не могу найти, там уже складываются в какие-то более длинные передачи и они уже по времени в микросекундах занимают время и мне что-то кажется что с этим самая большая проблема.
Самая большая проблема я не могу найти и не знаю как сделать так, чтоб можно было пометить на RX канале те моменты, когда я касаюсь сенсорной панели и определить сколько они по времени занимают (вернее нашел что-то типа 0xE0, 0xFF, 0x01, 0x02 и они повторяются чаще всего, видимо это что-то типа отметки начала и конца блоков передачи (если я правильно понимаю) и 0х00 как показатель того что в этот момент ничего не нажимается. Если можно  найти как-то отфильтровать, что типа не вот эти все описанные выше значения, а что-то кроме них, то может быть можно было бы явно найти, чтоб 100%но понять, что я в этим моменты нажимал?
Я нашел, что можно сохранить отдельно сам захваченный материал, чтоб открыть его потом отдельно в программе Logic 2 ( можно скачать отсюда https://www.saleae.com/ru/downloads/ чтоб можно было рассмотреть более детально и понять то, чего не понял я). Я пробовал, да, действительно открывается, странно, что я этого раньше не нашел, хотя до того делал экспорт и вам кидал вместе с картинками.
[Logic2captures.zip](https://github.com/r2axz/bluepill-serial-monster/files/7090544/Logic2captures.zip)

Может снова таки, если будут какие-то идеи как сделать как-то анализ более понятным, если вы самостоятельно взгляните - хотелось бы чтоб вы озвучили, мне просто действительно хочется понять все это дело :))))

steelpuxnastik, 2021-09-01:
В общем, произошел троллинг 80 уровня системы надо мной. А вас, как бы, задело вместе со мной :)
Собираю же систему, чтоб посмотреть все то, что описал выше, сначала конечно же проверил как с CP2102, все без проблем, включаю все в bluebill-serial-monster в надежде посмотреть на все те приколы, что теоретически  должны были вылезти и.. ничего не произошло: все работает ОТЛИЧНО :) 
Не помню, какая там версия прошивки стоит, эта или 2.4.2, но вся суть в том, что я изначально все это тестировал у себя на ноутбуке Acer 5560g, морально старый, но свою работу делает более чем исправно, и с игрой к тому же, также никаких проблем нет и ровно потому я его для тестирования и использовал и я даже подумать не мог что именно в нем самом кроется проблема. То есть, теперь проблемы по сути нет, она снята, так как целевым устройством не ноутбук является, а другой компьютер (проверил на двух компьютерах), где это будет использоваться, но вопрос теперь в другом - почему именно на ноутбуке эта проблема вылезла, особенно с учетом того, что на CP2102 без проблем? Этот момент как-то радикально сбил с толку.

steelpuxnastik, 2021-09-02:
Система Windows 10 21H1 последняя на данный момент вроде как, на одном из стационаров также и я не помню чтоб я что-то для него ставил когда подключал - само заработало вроде когда покдлючил, а на другом на целевом - Windows Embedded 8.1 и он сходу драйвер не нашел, потому скормил ему драйвер отсюда, с репозитория и тогда все заработало. По-моему на ноутбуке точно также ставил драйверы, так как само не заработало, но тут уже что не вспомню, слишком давно было.
[image]

r2axz, 2021-09-17:
Спасибо за информацию. К сожалению драйвер обычный. Поэтому, все равно остается загадкой что именно не взлетело с вашим ноутом. То ли софт, то ли какая-то особенность железа. Если есть возможность загрузить ноутбук с другой ОС (типа live CD или что-то еще), то было бы интересно проверить.

Если желания дальше экспериментировать нет, то надо решить что делать с багрепортом: закрываем или продолжаем исследовать...

r2axz, 2021-09-28:
This does not seem to be a bluepill-serial-monster issue. Closing due to inactivity. Please reopen if needed.

closed 2021-09-28 by r2axz