r2axz/bluepill-serial-monster #54: Добавить управление несколькими GPIO и чтение с  входов АЦП

kgkruteh4, 2022-03-15:
Если есть возможность просьба добавить возможность управление несколькими GPIO в режиме In/Out и чтение с  нескольких входов АЦП.

kgkruteh4, 2022-03-16:
Где-то в форумах читал сообщение от Вас, что вы хотели добавить аналогичный функционал.
Да, основное назначение конвертер USB2Serial. То, что я предлагаю добавить, это дополнение функционала еще до USB<->GPIO/ADC.

kgkruteh4, 2022-03-16:
Спасибо за развернутый ответ.
Наверно Вы правы и мой запрос слишком специфичен.
И он не совсем вписывается в код устройства.
Запрос снимается.

r2axz, 2022-03-16:
Спасибо за понимание, в любом случае я буду иметь в виду что такая функциональность нужна некоторой части пользователей (и их теперь как минимум двое) и, возможно, в будущем пересмотрю планы. Но в любом случае, мне надо доделать сначала звук. Основная проблема в том, что у меня нет в голове концепции того как это лучше сделать и почему именно так. Ну и, конечно, ограниченный ресурс -- это некоторая объективная проблема.

closed 2022-03-16 by r2axz