r2axz/bluepill-serial-monster #55: Suggestion: Blackpill support?

hanetzer, 2022-03-31:
Its my understanding that the two chips are pretty close together in features and capabilities.
Would be fun to extend this to support both, if possible.

r2axz, 2022-03-31:
Sounds like a good idea. I need to learn more about the blackpill peripherals and I guess order one, since I don't have any :)

hanetzer, 2022-03-31:
I do own a few, so if you could point out key bits I could perhaps do the code and PR it to ya

hanetzer, 2022-03-31:
well, git grep is showing me things. btw, where does stm32f1xx.h come from?
its not in the repository.

hanetzer, 2022-03-31:
Hrm. Building it for stm32f1xx gives the following error:
```
mkdir -p build
arm-none-eabi-gcc -MT build/usb_descriptors.o -MMD -MP -MF build/usb_descriptors.d -DSTM32F103xB -DHSE_VALUE=8000000U -I/home/hanetzer/Firmware/stm32/STM32CubeF1/Drivers/CMSIS/Core/Include -I/home/hanetzer/Firmware/stm32/STM32CubeF1/Drivers/CMSIS/Core_A/Include -I/home/hanetzer/Firmware/stm32/STM32CubeF1/Drivers/CMSIS/Device/ST/STM32F1xx/Include -mthumb -mcpu=cortex-m3 -Wall -O3 -ggdb  -DDEVICE_VERSION_MAJOR=latest -c usb_descriptors.c -o build/usb_descriptors.o
In file included from usb_core.h:13,
                 from usb_descriptors.c:7:
<command-line>: error: ‘latest’ undeclared here (not in a function)
usb_std.h:67:44: note: in definition of macro ‘USB_BCD_VERSION’
   67 | #define USB_BCD_VERSION(maj, min, rev)  (((maj & 0xFF) << 8) | ((min & 0x0F) << 4) | (rev & 0x0F))
      |                                            ^~~
usb_descriptors.c:99:43: note: in expansion of macro ‘DEVICE_VERSION_MAJOR’
   99 |     .bcdDevice          = USB_BCD_VERSION(DEVICE_VERSION_MAJOR, DEVICE_VERSION_MINOR, DEVICE_VERSION_REVISION),
      |                                           ^~~~~~~~~~~~~~~~~~~~
make: *** [Makefile:78: build/usb_descriptors.o] Error 1
```

hanetzer, 2022-03-31:
Ahh. Its a problem with the GIT_VERSION macro or the like. `make GIT_VERSION=1` works (well, it builds at least)

hanetzer, 2022-03-31:
Hmm. The code could be refactored in a manner similar to https://github.com/blackmagic-debug/blackmagic and handle
the platform specifics in some subdirs.

r2axz, 2022-04-01:
> btw, where does stm32f1xx.h come from?

This comes from CMSIS, I use https://github.com/STMicroelectronics/STM32CubeF1.git, but it looks like you've figured that out already.

> Hrm. Building it for stm32f1xx gives the following error:
> 
> ```
> mkdir -p build
> ...

That's a bug. I didn't take into account the "latest" case. Will fix that.

> Hmm. The code could be refactored in a manner similar to https://github.com/blackmagic-debug/blackmagic and handle
> the platform specifics in some subdirs.

Well, probably yes. Another possibility is to use ifdefs, which may be easier if we are only interested in two platforms at the moment. However, if someday we need to support more platforms in addition to black/bluepill, it'll come to this anyways.