r2axz/bluepill-serial-monster #56: Windows compilation

Kabron287, 2022-04-26:
All this of course is very good, but is there a way to compile it in Windows, preferable with known instruments: Cube, Keil, IAR...?

Kabron287, 2022-04-26:
Well, I tryed to compile With CubeIDE as MakeFileProject and got sevral errors:
usb_descriptors.c:96:27: error: initializer element is not constant
     .bMaxPacketSize     = usb_endpoints[usb_endpoint_address_control].rx_size,

Kabron287, 2022-04-26:
When I changed:
  //      .wMaxPacketSize         = usb_endpoints[usb_endpoint_address_cdc_0_data].tx_size,
		.wMaxPacketSize         = 16,

Compilation was success

r2axz, 2022-04-26:
Hi,

Thanks for this issue. My bad. I used non-const initializes in a few places in the code. Although recent versions of gcc and clang can live with that, this violates the C standard. Older versions of gcc (which you probably use with you installation of CubeIDE) consider this an error. Anyhow, fixed in #57, please test it should work with CubeIDE now.

As for Keil and IAR, I don't think I care for those at this point.

closed 2022-05-11 by r2axz