r2axz/bluepill-serial-monster #61: [info] max baud rate

yashikada, 2022-09-13:
what is the maximum baud rate you have achieved on a uart?
I would like to avoid buying an expensive ftdi ft232h.

Thank you

r2axz, 2022-09-13:
Please see README.md

>> Supports all baud rates up to 2 MBaud;

I have no idea what baud rate you need, but ft232h is notably faster.

closed 2022-09-15 by r2axz

yashikada, 2022-09-17:
Thank you @r2axz for informations and sorry for the delay.
Your README has many informations, so it's not easy read all.
In this case baud is a bit, rigth? So max speed is 2 Mbit/s while a ft232h reaches 12Mbaud (UART) up to 40MB/s (Sync FIFO).
2 Mbit/s is a good speed for me.

r2axz, 2022-09-17:
> In this case baud is a bit, rigth? So max speed is 2 Mbit/s while a ft232h reaches 12Mbaud (UART) up to 40MB/s (Sync FIFO).
2 Mbit/s is a good speed for me.

Yes, this is correct.