r2axz/bluepill-serial-monster #62: [info] communication with device powered at 5V

yashikada, 2022-09-17:
Sorry for my newbie question, I'm not electrical engineer.
From README

**When configured as an output, none of the STM32F103C8T6 pins is 5 V tolerant. Make sure you don't accidentally get more than 4.0 V on such pin or damage may occur.**

If I connect TX of BluePill to pin RX of a device powered at 5V, can I broken the STM32F103C8T6? RX should be high impedance or doesn't matter?
Other connection should be ok because RX on bluepill is 5V tolerant except for UART2.

I used bluepill for [serialudpi](https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md) where RX and TX are shorted with a resistor and a diode.
I powered the MCU at 3.3V and works no damage, but if I power the MCU at 5V damage may occur?

r2axz, 2022-09-17:
> When configured as an output, none of the STM32F103C8T6 pins is 5 V tolerant. Make sure you don't accidentally get more than 4.0 V on such pin or damage may occur.

The device to which you connect an output of bluepill-serial-monster may have a [pull-up resistor](https://en.wikipedia.org/wiki/Pull-up_resistor) that ties the device input to a 5V power supply rail. This pull-up resistor may be an actual resistor component or an internal pull-up of the IC in that device. If that is the case, a damage to STM32F103C8T6 may occur. Anything below 4V is fine.

> If I connect TX of BluePill to pin RX of a device powered at 5V

The fact the device is powered at 5V does not necessarily mean its inputs are tied to the 5V power supply rail.

> RX should be high impedance or doesn't matter?

That depends on the device. The outputs on stm32f103c8t6 may source or sink up to 25 mA of current, so they can be connected to a relatively low impedance outputs. Normally, logic inputs have much higher impedance than stm32f103c8t6 outputs can deal with, so this is not a problem.

> Other connection should be ok because RX on bluepill is 5V tolerant except for UART2.

Correct.

> I powered the MCU at 3.3V and works no damage, but if I power the MCU at 5V damage may occur?

I cannot guarantee anything, but based on [serialudpi](https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md) documentation everything should be fine if your MCU is powered at 5 V. Again, never tried this myself so please don't quote me on this :)

yashikada, 2022-09-18:
Very great reply

> I cannot guarantee anything, but based on [serialudpi](https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md) documentation everything should be fine if your MCU is powered at 5 V. Again, never tried this myself so please don't quote me on this :)

No problem. A day I will try with VCC 5V. 
For now my scope was program a MCU with UDPI without buy a programmer or a ft232, mission accomplished so coffee is guarantee.

closed 2022-09-18 by yashikada

r2axz, 2022-09-18:
Thank you and good luck with your project!